WebThingsIO/webthing-arduino #1: Provide pointer for Add Thing by URL feature

rei-vilo, 2018-02-18:
Could you please provide a pointer to the **Add Thing by URL** feature?

Thank you!

rei-vilo, 2018-02-18:
Connecting a browser directly to the ESP8266 displays

```
{
 "name": "ESP8266 LED",
  "type": "onOffSwitch",
  "href": "/",
  "properties": {
    "on": {
      "type": "boolean",
      "href": "/properties/on"
    }
  },
  "actions": {},
  "events": {}
}
```

hobinjk, 2018-02-18:
The Add Thing by URL feature doesn't exist yet but I plan to mock up an initial implementation for it soon. Until then, its tracking issue is https://github.com/mozilla-iot/gateway/issues/69. Sorry for the confusion!

julienrat, 2018-02-19:
Hi ! Same for me ! impatient to have this feature ! ;-)

rei-vilo, 2018-03-17:
Does [release 0.3.1](https://github.com/mozilla-iot/gateway/releases/tag/0.3.1) include the `Add Thing by URL` feature?

hobinjk, 2018-04-11:
Add Thing by URL is present on the current master branch of the Gateway and will be available in our upcoming 0.4.0 release. Thanks for waiting!

closed 2018-04-11 by hobinjk