WebThingsIO/webthing-arduino #2: Adding ESP8266 to Mozilla IoT

mcprakash, 2018-04-15:
Hi @hobinjk
 I am stuck at adding the device to my Gateway. The ESP8266 (12 E) connects to my internet and I am able to ping to the device. 

 I did a manual installation of the Mozilla-IoT on my R-Pi as described here -> https://github.com/mozilla-iot/gateway

 This also happened with a Microchip device (that was tested at the Mozilla IoT workshop) that the Gateway is not able to recognize the IoT device. 

 PFA an image of the same. 
[image]

 Please help me out on this. 

Best
Chandra

hobinjk, 2018-04-15:
Hi Chandra,
Sorry that you're encountering difficulties! Can you try running `ping esp8266.local` from the command line?

It should also be possible to use "Add by URL" with "http://192.168.1.11" if discovery continues to fail. If you could also share the contents of ~/.mozilla-iot/log/run-app.log from the Raspberry Pi that would help with debugging the discovery process.

mcprakash, 2018-04-15:
Hi James
 PFA the output of ping from my computer connected to the same network

[image]

Tried adding using "Add by URL" with "http://192.168.1.11" and got "Device Fetch Failed"

Best
Chandra

hobinjk, 2018-04-15:
Are you sure that your ESP8266 and Pi are on the same network? If the Add by URL with an IP address isn't working then they're somehow unable to contact eachother. Can you make sure that opening http://192.168.1.11 in your web browser works? It should have a JSON list of Thing descriptions. If your ESP and Pi and are on the same network and you can see the list of Thing descriptions then the most like cause of the problem is your router trying to block communication between the two devices.

mcprakash, 2018-04-15:
Hi James
 Yes, opening it in my browser works

 PFA an image of the same.

 I think the Pi was attached using the Ethernet while the ESP was attached
using the WiFi. Should not make any difference though.

 Any ideas of how I can resolve my router my blocking the communication
like disabling security or so. I am using the TP-LINK TC-W7960 modem
router.

 Thanks again for all the help.

Best
Chandra

hobinjk, 2018-04-16:
I think the best avenue to test now would be to run `ping 192..168.1.11` on the Pi itself. You can do this by logging into the Pi using these instructions https://github.com/mozilla-iot/wiki/wiki/Logging-into-the-Raspberry-Pi.

You can also try modifying your router's settings. I don't know what the specific setting for that model is but usually if you disable "DoS protection" or an equivalent setting you should be good to go. In general, any setting that interferes with mDNS, Bonjour, or communicating between wired and wireless devices is one that you want to take a look at.

It's also possible that all of the problems will magically disappear if you connect your Pi wirelessly. The instructions for doing that manually are on the Logging into the Raspberry PI page linked above.

mcprakash, 2018-04-16:
Hi James, 
 I am connecting the R-Pi wireless (from yesterday night). Yes, ping to the device works on the Pi. 
 No luck so far. 

 I ran npm test and got a few tests that failed. 
 PFA the debug log for the tests 
[2018-04-16T07_26_23_342Z-debug.log](https://github.com/mozilla-iot/webthing-esp8266/files/1917073/2018-04-16T07_26_23_342Z-debug.log)

 Please help me out. If you have RealVNC, I can provide you credential access to my pi, so you can take a look. 

Best
Chandra

mcprakash, 2018-04-16:
Hi James
 Thanks for all the support. I finally got it working. Restarted my pi a couple of times and finally it found the device. Not sure what went wrong in the first place but am glad that it is finally working. 

[image]

Best
Chandra

hobinjk, 2018-04-16:
Glad to hear it! It's definitely strange that it wasn't working earlier.