WebThingsIO/webthing-arduino #3: Compilation error

bphermansson, 2018-04-17:
I try to compile the LED example but it doesn't work:

```
/Arduino/sketchbook/libraries/webthing-esp8266-master/WebThingAdapter.h:42:8: error: no matching function for call to 'ESP8266WebServer::on(String&, WebThingAdapter::begin()::__lambda1)'
       });
```

What am I missing?

hobinjk, 2018-04-17:
I'd make sure that you have the ESP8266 support installed which is documented at https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/using-arduino-ide. You may also need to manually install ESP8266WebServer if it isn't listed under Sketch > Include Library

bphermansson, 2018-04-17:
I have those installed as I built several Esp-based things before. The HelloServer from the Examples compiles without errors.

hobinjk, 2018-04-17:
Okay, that narrows it down significantly. Sorry that you're encountering issues!

What version of the Arduino IDE are you using? It looks like it's confused about using a lambda as the ESP8266WebServer's void callback but that works on my computer with Arduino 1.8.5.

bphermansson, 2018-04-17:
I have 1.8.5 too.

hobinjk, 2018-04-17:
I'm not sure why adding the capture of a variable would change the resolution of the lambda's type like that. Could you try replacing your local copy of WebThingAdapter.h with https://gist.github.com/hobinjk/7539d90edfa13c4cb2e230f3d3856536? It's explicit about the function type and should be a bit more likely to resolve correctly.

bphermansson, 2018-04-18:
No luck with that either

```
libraries/webthing-esp8266-master/WebThingAdapter.h:43:40: error: no matching function for call to 'ESP8266WebServer::on(String&, std::function<void()>&)'
       server.on(deviceBase, baseHandler);
                                        ^
```

bphermansson, 2018-04-19:
@OlavGullaksen: It doesn't work for me. I don't get this...

@hobinjk 
```
In file included from /media/nasFiler/Programmering/Arduino/sketchbook/Esp/LED/LED.ino:13:0:
/tmp/arduino_build_371723/sketch/WebThingAdapter.h: In member function 'void WebThingAdapter::begin()':
WebThingAdapter.h:43: error: no matching function for call to 'ESP8266WebServer::on(String&, std::function<void()>&)'
       server.on(deviceBase, baseHandler);
                                        ^
/tmp/arduino_build_371723/sketch/WebThingAdapter.h:43:40: note: candidates are:
In file included from /tmp/arduino_build_371723/sketch/WebThingAdapter.h:15:0,
                 from /media/nasFiler/Programmering/Arduino/sketchbook/Esp/LED/LED.ino:13:
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, THandlerFunction handler);
        ^
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note:   no known conversion for argument 1 from 'String' to 'const char*'
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn);
        ^
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note:   candidate expects 3 arguments, 2 provided
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn);
        ^
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note:   candidate expects 4 arguments, 2 provided
In file included from /media/nasFiler/Programmering/Arduino/sketchbook/Esp/LED/LED.ino:13:0:
WebThingAdapter.h:52: error: no matching function for call to 'ESP8266WebServer::on(String&, HTTPMethod, std::function<void()>&)'
         server.on(propertyBase, HTTP_GET, propertyGetHandler);
                                                             ^
/tmp/arduino_build_371723/sketch/WebThingAdapter.h:52:61: note: candidates are:
In file included from /tmp/arduino_build_371723/sketch/WebThingAdapter.h:15:0,
                 from /media/nasFiler/Programmering/Arduino/sketchbook/Esp/LED/LED.ino:13:
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, THandlerFunction handler);
        ^
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note:   candidate expects 2 arguments, 3 provided
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn);
        ^
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note:   no known conversion for argument 1 from 'String' to 'const char*'
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn);
        ^
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note:   candidate expects 4 arguments, 3 provided
In file included from /media/nasFiler/Programmering/Arduino/sketchbook/Esp/LED/LED.ino:13:0:
WebThingAdapter.h:57: error: no matching function for call to 'ESP8266WebServer::on(String&, HTTPMethod, std::function<void()>&)'
         server.on(propertyBase, HTTP_PUT, propertyPutHandler);
                                                             ^
/tmp/arduino_build_371723/sketch/WebThingAdapter.h:57:61: note: candidates are:
In file included from /tmp/arduino_build_371723/sketch/WebThingAdapter.h:15:0,
                 from /media/nasFiler/Programmering/Arduino/sketchbook/Esp/LED/LED.ino:13:
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, THandlerFunction handler);
        ^
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note:   candidate expects 2 arguments, 3 provided
/home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)
[... cut ...]
Using library ArduinoJson at version 5.13.0 in folder: /media/nasFiler/Programmering/Arduino/sketchbook/libraries/ArduinoJson 
Using library ESP8266WebServer at version 1.0 in folder: /home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer 
Using library ESP8266mDNS in folder: /home/patrik/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS (legacy)
exit status 1
no matching function for call to 'ESP8266WebServer::on(String&, std::function<void()>&)'

```

bphermansson, 2018-04-21:
@hobinjk That was it! It compiles fine after upgrading the libraries. Thanks!

closed 2018-05-30 by hobinjk

Shayantika-Dhar, 2024-01-16:
/home/xlayer/Arduino/esp32-webthings/esp32-webthings.ino:4:10: fatal error: WebThingsAdapter.h: No such file or directory
 #include <WebThingsAdapter.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.

having error at webthingsadapter. dont know where to add