WebThingsIO/webthing-arduino #7: Add Activity and Event methods to Thing object

phatpenguintn, 2018-06-04:
The ESP8266 libraries for creating web things work with well defined things like lights, but do not allow things like creating a garage door opener thing which requires actions and events in order to maintain states etc.

rafspiny, 2018-07-16:
@mrstegeman Correct me if this is not related. 
I was wondering: how many libraries thatare to validate a JSON? Found a couple of solution, but not for the Arduino framework.
- https://github.com/tristanpenman/valijson
- https://github.com/Tencent/rapidjson

From what I understood, JSON Schema validation is not provided by ArduinoJSON atm.

I played with the library and added Action to the thing object, with the generation of the links and so on. But of course, without validation.

Just for some context: I built a thing with a soil moisture sensor and a water pump, wired them up to a ESP8266, and wrote a JSON-LD for it (unusable at the moment). Actions are implemented as properties at the moment.

Thanks.

lgLindstrom, 2019-03-08:
Any update/plans on this ??

commit 1065e0b6 (2020-01-06): Add full action support.

Fixes #7
closed 2020-01-06 by mrstegeman