WebThingsIO/webthing-arduino #10: Devices number is limited

nledez, 2018-06-12:
I try to add devices, all works fine.
But with a GET on / is limited by ESP memory.

I my first version I have full description on each device.

I create on more concise, but only one more :/

```
[
  {
    "name": "Bed lights",
    "type": "onOffSwitch",
    "href": "/things/bed",
    "properties": {
      "on": {
        "type": "boolean",
        "href": "/things/bed/properties/on"
      }
    }
  },
  {
    "name": "Lamp left",
    "type": "onOffSwitch",
    "href": "/things/brLbL",
    "properties": {
      "on": {
        "type": "boolean",
        "href": "/things/brLbL/properties/on"
      }
    }
  },
  {
    "name": "Lamp right",
    "type": "onOffSwitch",
    "href": "/things/brLbR",
    "properties": {
      "on": {
        "type": "boolean",
        "href": "/things/brLbR/properties/on"
      }
    }
  },
  {
    "name": "Standing lamp",
    "type": "onOffSwitch",
    "href": "/things/lamp",
    "properties": {
      "on": {
        "type": "boolean",
        "href": "/things/lamp/properties/on"
      }
    }
  },
  {
    "name": "Desk lamp",
    "type": "onOffSwitch",
    "href": "/things/dskL",
    "properties": {}
  }
]
```

With code:
```
#include <ESP8266WebThingAdapter.h>
#include <WebThingAdapter.h>
#include <Thing.h>
#include <WiFi101WebThingAdapter.h>
#include <ESP32WebThingAdapter.h>
#include <ArduinoOTA.h>
#include <RCSwitch.h>

// ======= RCSwitch
RCSwitch mySwitch = RCSwitch();
#define pinRF D1

// ======= Wifi
const char* ssid = "yes it's my SSID!";
const char* password = "And my pass too";

// ======= Webthings
#ifdef ESP32
ESP32WebThingAdapter adapter("esp32");
#endif
#ifdef ESP8266
ESP8266WebThingAdapter adapter("esp8266");
#endif

ThingDevice bed("bed", "Bed lights", "onOffSwitch");
ThingProperty bedOn("on", "When bed is on", BOOLEAN);
bool bedLastOn = false;

ThingDevice brLbL("brLbL", "Lamp left", "onOffSwitch");
ThingProperty brLbLOn("on", "When lamp left is on", BOOLEAN);
bool brLbLLastOn = false;

ThingDevice brLbR("brLbR", "Lamp right", "onOffSwitch");
ThingProperty brLbROn("on", "When lamp right is on", BOOLEAN);
bool brLbRLastOn = false;

ThingDevice lamp("lamp", "Standing lamp", "onOffSwitch");
ThingProperty lampOn("on", "When standing lamp is on", BOOLEAN);
bool lampLastOn = false;

ThingDevice dskL("dskL", "Desk lamp", "onOffSwitch");
ThingProperty dskLOn("on", "When desk lamp is on", BOOLEAN);
bool dskLLastOn = false;

ThingDevice lrLampF("lrLampF", "Front bubble light", "onOffSwitch");
ThingProperty lrLampFOn("on", "When front bubble light is on", BOOLEAN);
bool lrLampFLastOn = false;

ThingDevice lrLampR("lrLampR", "Rear bubble light", "onOffSwitch");
ThingProperty lrLampROn("on", "When rear bubble light is on", BOOLEAN);
bool lrLampRLastOn = false;

ThingDevice tv("tv", "TV", "onOffSwitch");
ThingProperty tvOn("on", "When TV is on", BOOLEAN);
bool tvLastOn = false;

void setupAdapter() {
  mySwitch.enableTransmit(pinRF);
  mySwitch.setProtocol(1);
  mySwitch.setPulseLength(149);
  mySwitch.setRepeatTransmit(7);

  bed.addProperty(&bedOn);
  adapter.addDevice(&bed);
  
  brLbL.addProperty(&brLbLOn);
  adapter.addDevice(&brLbL);
  
  brLbR.addProperty(&brLbROn);
  adapter.addDevice(&brLbR);
  
  lamp.addProperty(&lampOn);
  adapter.addDevice(&lamp);
  
  dskL.addProperty(&dskLOn);
  adapter.addDevice(&dskL);
  
  lrLampF.addProperty(&lrLampFOn);
  adapter.addDevice(&lrLampF);
  
  lrLampR.addProperty(&lrLampROn);
  adapter.addDevice(&lrLampR);
  
  tv.addProperty(&tvOn);
  adapter.addDevice(&tv);
  
  adapter.begin();
  Serial.println("HTTP server started");
}

void handleAdapter() {
  adapter.update();
  // One thing
  if (lampOn.getValue().boolean) {
    if (!lampLastOn) {
      Serial.println("Switch on lamp");
      mySwitch.send("01");
      lampLastOn = true;
    }
  } else {
    if (lampLastOn) {
      Serial.println("Switch off lamp");
      mySwitch.send("01");
      lampLastOn = false;
    }
  }

  // One thing
  if (dskLOn.getValue().boolean) {
    if (!dskLLastOn) {
      Serial.println("Switch on dskL");
      mySwitch.send("01");
      dskLLastOn = true;
    }
  } else {
    if (dskLLastOn) {
      Serial.println("Switch off dskL");
      mySwitch.send("01");
      dskLLastOn = false;
    }
  }

  // One thing
  if (lrLampFOn.getValue().boolean) {
    if (!lrLampFLastOn) {
      Serial.println("Switch on lrLampF");
      mySwitch.send("01");
      lrLampFLastOn = true;
    }
  } else {
    if (lrLampFLastOn) {
      Serial.println("Switch off lrLampF");
      mySwitch.send("01");
      lrLampFLastOn = false;
    }
  }

  // One thing
  if (lrLampROn.getValue().boolean) {
    if (!lrLampRLastOn) {
      Serial.println("Switch on lrLampR");
      mySwitch.send("01");
      lrLampRLastOn = true;
    }
  } else {
    if (lrLampRLastOn) {
      Serial.println("Switch off lrLampR");
      mySwitch.send("01");
      lrLampRLastOn = false;
    }
  }

  // One thing
  if (tvOn.getValue().boolean) {
    if (!tvLastOn) {
      Serial.println("Switch on tv");
      mySwitch.send("01");
      tvLastOn = true;
    }
  } else {
    if (tvLastOn) {
      Serial.println("Switch off tv");
      mySwitch.send("01");
      tvLastOn = false;
    }
  }

  // One thing
  if (bedOn.getValue().boolean) {
    if (!bedLastOn) {
      Serial.println("Switch on bed");
      mySwitch.send("01");
      bedLastOn = true;
    }
  } else {
    if (bedLastOn) {
      Serial.println("Switch off bed");
      mySwitch.send("01");
      bedLastOn = false;
    }
  }

  // One thing
  if (brLbLOn.getValue().boolean) {
    if (!brLbLLastOn) {
      Serial.println("Switch on brLbL");
      mySwitch.send("01");
      brLbLLastOn = true;
    }
  } else {
    if (brLbLLastOn) {
      Serial.println("Switch off brLbL");
      mySwitch.send("01");
      brLbLLastOn = false;
    }
  }

  // One thing
  if (brLbROn.getValue().boolean) {
    if (!brLbRLastOn) {
      Serial.println("Switch on brLbR");
      mySwitch.send("01");
      brLbRLastOn = true;
    }
  } else {
    if (brLbRLastOn) {
      Serial.println("Switch off brLbR");
      mySwitch.send("01");
      brLbRLastOn = false;
    }
  }

}

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }
  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  setupAdapter();
}

void loop() {
  ArduinoOTA.handle();
  handleAdapter();
}
```

rzr, 2018-06-16:
And are only 2 items working fine ?

rzr, 2018-06-18:
Yes I was surprised that we would get out of ESP8266 resources, if so ESP32 could also been considered . but optimization is a better bet.

nledez, 2018-06-20:
@rzr is it's works fine with devices correctly show.

@hobinjk thanks for idea, I patch with https://github.com/nledez/webthing-esp8266/commit/f073b506b42178238f6515390ca4faa3ca4d82a5 and curl works fine :)

nledez, 2018-06-20:
And now works in gateway \o/

[image]

hobinjk, 2018-06-20:
Excellent! I wrote a patch which did a fancy streaming solution but it didn't save a sizable amount of memory in the grand scheme of things. I'm just going to uplift your changes, thanks for figuring it out!

closed 2018-06-20 by hobinjk