WebThingsIO/webthing-arduino #14: New ArduinoJSON beta doesn't compile with ESP8266

markbeee, 2018-06-23:
From version 6.0.0 (beta) Arduino JSON library doesn't compile with webthing-arduino/ ESP8266.

Quote from [https://github.com/bblanchon/ArduinoJson/issues/760] 
"Indeed ArduinoJson 6 replaced the concept of JsonBuffer with the concept of JsonDocument.
To fix this error, you need either:
Downgrade ArduinoJson to 5.x (recommended as v6 is still in beta)
Upgrade your program to work with ArduinoJson 6"

hobinjk, 2018-07-09:
According to the ArduinoJson website version 6 will be in beta for the rest of 2018 so I think the correct option is to require the use of version 5.

rzr, 2018-08-11:
Please see if this can be tweaked in make files I just added.

willdurand, 2019-03-18:
@hobinjk the work to upgrade from ArduinoJson 5 to 6 does not look huge: https://arduinojson.org/v6/doc/upgrade/. Would it make sense to upgrade?

Supporting both versions would be doable, but would it make sense?

matiasdelellis, 2019-07-24:
Hi, it's already stable, so IMHO is a good time to update this library.. :smiley: 

 * https://arduinojson.org/2019/03/05/version-6-9-0/

Thanks for all.

matiasdelellis, 2019-10-21:
@markbeee @rzr @willdurand @novski 
Sorry to mention you with @ :sweat_smile: 

@hobinjk is working on this in the pull request #95 
.... and needs to be tested.  :grimacing: 

Thanks and regards,
Matias,

rzr, 2019-10-21:
If you update it please also try other sample apps built with top makefile

rzr, 2019-11-28:
Relate-to: https://github.com/mozilla-iot/webthing-arduino/pull/95

PR #99 (2019-12-30): ArduinoJson 6
Fixes #14
closed 2019-12-30 by mrstegeman