WebThingsIO/webthing-arduino #21: Missing included file <ArduinoMDNS.h>

abbydillon, 2018-07-15:
The library **WiFi101WebThingAdapter.h** includes a file called **ArduinoMDNS.h**, but this file is not in the repo and is not available on Arduino IDE Library manager.  A quick web search does not turn up a definitive source. 

Could you either include it in the repo, link to it elsewhere, or remove it if it's not needed? Thank you. 

I'm trying to figure out why the Gateway does not find my device, and thinking this could be part of the problem. Thanks.

mone27, 2018-07-15:
Have you installed the native-web-thing addon  on the gateway? Anyway you can check that your webthing is working simply connecting with yours browser at the webthing  http server.

hobinjk, 2018-07-16:
The ArduinoMDNS library is https://github.com/arduino-libraries/ArduinoMDNS which should be included by default as far as I know (it's installed in my local Arduino IDE 1.8.5).

I was having some issues with getting the mDNS library to create properly-formatted packets but a board running the latest version of this software and of thing-url-adapter (0.2.1) should function correctly. You can verify the version of thing-url-adapter from the gateway's settings > addons page

rzr, 2018-07-25:
FYI I have hacked some makefiles to rebuild examples without using IDE... would that help ?

https://github.com/TizenTeam/webthing-esp8266/tree/master/examples/LevelSensor

So47, 2018-09-21:
Have you solved it ?

hobinjk, 2018-09-21:
I think this is solved either by setting up the Arduino IDE so it recognizes its built-in libraries or by using the provided makefiles. Discovery can still easily be broken by network or chip issues.

closed 2018-09-21 by hobinjk

So47, 2018-09-23:
When I added arduinoMDNS library , it gives me another error, 

Sketch uses 49140 bytes (152%) of program storage space. Maximum is 32256 bytes.

Global variables use 1784 bytes (87%) of dynamic memory, leaving 264 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board Arduino/Genuino Uno.

hobinjk, 2018-09-25:
@So47 Unfortunately this library includes a lot of other dependencies like ArduinoJson and WiFi101 which make it too large for the Uno and some other boards. The board I use is a Mega which does have enough space available.

rzr, 2018-09-25:
mDNS could be optional to target constrained devices...

hobinjk, 2018-09-25:
If I remember correctly from the last time I looked at this the JSON encoder/decoder is the largest contributor but even with minimal encoding/decoding webthing-arduino still couldn't fit in the 32k of the Uno.

For very constrained devices something like https://github.com/bhagman/usb-serial-adapter is probably more suitable.

I do think there's a bunch of room for optimizing webthing-arduino and making mDNS easy to remove would be a nice win.

freaktechnik, 2021-07-23:
I think it would make sense to provide an ifdef to skip all usage of the mDNS infrastructure, since it appears the library isn't available for all devices that might for example support WiFi101.