WebThingsIO/webthing-arduino #29: Esp crash

msomjit, 2018-08-19:
Hi, I've uploaded che attached sketch (lamp demo) into my ESP8266 but every time I press the "+" button to discovery the "thing", ESP crash and restart several times. I can discovery  manually typing "http://192.168.0.26/things/led" via "Add by URL". and  It all runs smoothly once the thing is discovered, but if i hit "+" again, the Esp crashes again.

Serial  port log:
........
Connected to WiFiM
IP address: 192.168.0.26
MDNS responder started
HTTP server started
http://192.168.0.26/things/led
**<---- i hit "+" button**
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

Connecting to "WiFiM"

.....
Connected to WiFiM
IP address: 192.168.0.26
MDNS responder started
HTTP server started
http://192.168.0.26/things/led
**<---- "Add by URL"**
led: 1
led: 0
led: 1
led: 0
led: 1
**<---- i hit "+" button again **
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

Connecting to "WiFiM"

.....
Connected to WiFiM
IP address: 192.168.0.26
MDNS responder started
HTTP server started
http://192.168.0.26/things/led
led: 1
led: 0

**<---- i hit "+" button again **
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

Connecting to "WiFiM"

.....
Connected to WiFiM
IP address: 192.168.0.26
MDNS responder started
HTTP server started
http://192.168.0.26/things/led
led: 1
led: 0
**<---- i hit "+" button again **
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

Connecting to "WiFiM"

.....
Connected to WiFiM
IP address: 192.168.0.26
MDNS responder started
HTTP server started
http://192.168.0.26/things/led
led: 1
**<---- i hit "+" button again **
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

Connecting to "WiFiM"

.....
Connected to WiFiM
IP address: 192.168.0.26
MDNS responder started
HTTP server started
http://192.168.0.26/things/led

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

Connecting to "WiFiM"

.....
Connected to WiFiM
IP address: 192.168.0.26
MDNS responder started
HTTP server started
http://192.168.0.26/things/led
led: 1
led: 0
led: 1
led: 0
led: 1
led: 0
led: 1

[Test.zip](https://github.com/mozilla-iot/webthing-arduino/files/2300517/Test.zip)

msomjit, 2018-08-25:
I've bought an ESP23 and loaded the same sketch: all runs well, no issue.

download13, 2018-10-29:
I'm not sure if this is the same problem, but I kept getting a similar crash every time I requested `/` from the ESP8266.

I ended up fixing it by changing [line 128](https://github.com/mozilla-iot/webthing-arduino/blob/master/ESPWebThingAdapter.h#L128) to be a `DynamicJsonBuffer`. I think I read somewhere that the stack on the ESP8266 is limited to 4KB so the `StaticJsonBuffer` of that size might be causing problems.

mrstegeman, 2019-12-30:
Seems like this has probably been resolved with JSON fixes.

closed 2019-12-30 by mrstegeman