WebThingsIO/webthing-arduino #32: WiFi101: mDNS packet truncated at 64 byte mark

hobinjk, 2018-10-03:
Undefining LIMITED_RAM_DEVICE to get a big UDP buffer didn't help so it might be at the ArduinoMDNS layer instead

@kgiori This is the mystery bug which prevents discovery from working on WiFi101 devices.

kgiori, 2018-10-10:
@dhylands earlier quick hack that he made for a demo, required a gateway add-on but the mDNS part worked when running on the WINC1500 module:
https://github.com/mozilla-iot/http-on-off-wifi101

hobinjk, 2018-10-11:
Unfortunately that's the same approach used as the baseline here. The point where it breaks down is advertising the `webthing` mDNS service which runs into the WiFi101-specific 64 byte limit

mrstegeman, 2020-01-07:
@kgiori Do you know if this bug still exists? I don't have any WiFi101 hardware to test with.

kgiori, 2020-01-09:
I'll try to get to testing this tomorrow. I do have a MKR1000 that would
use the WiFi101 library.
kathy