WebThingsIO/webthing-arduino #33: MKR1010: Replace WiFi101 with WiFiNINA

Nicokeuk, 2018-10-09:
I tried the LED example with an Arduino MKR1010, and it wasn't connecting to my WiFi network. I had to change the WiFi101 library by the WiFiNINA library in the includes of WiFi101WebThingAdapter.h.
With this WiFiNINA library it worked directly.

`//#include <WiFi101.h>`
`#include <WiFiNINA.h>`

rzr, 2019-01-04:
why not using #ifdef s  ?

hobinjk, 2019-01-04:
:+1:, if there's a define like ARDUINO_BOARD_MKR1010 or something like that where we can filter based on whether we're compiling for an MKR1010 that would be great. Unfortunately I don't have access to an MKR1010 so I can't test it out locally.

Nicokeuk, 2019-03-14:
I couldn't find a parameter that would work with an #ifdef, but if you have any idea, I'd be happy to try them.

rzr, 2019-03-14:
at worst case you create one (ex: CONFIG_WANT_NET_WIFI_NINA etc) and you enable in build env ?

commit 70ca3f26 (2019-12-30): Support MKR1010.

Fixes #33
closed 2019-12-30 by mrstegeman