WebThingsIO/webthing-arduino #35: Using multiple ESP32 "Things" - it looks like last one added to gateway is the only one working correctly

ianarchbell, 2018-10-29:
I have a a set of ESP32-based examples running on our oddWires boards here:

https://github.com/iot-bus/iot-bus-mozilla-iot-examples

They all work fine individually and the library was easy to use. However, running more than one at the same time is turning out to be problematical. I do not know whether I am using the library incorrectly or whether there is an issue either in the library or gateway. For example. If I add a relay board I can happily turn it on or off using the moz-iot interface and I can turn it on/off locally using a touchscreen interface and the associated property is updated correctly in the interface. Once I add a PIR sensor or anything else I cannot control it from the moz-iot interface.

Running gateway in debug mode the gateway output is:

[image]

Unexpected end of JSON input often means there is none.  However, I don't know whether there is malformed JSON on the adapter. It would be useful if what was there was logged.

I'd appreciate a pointer on where to go. If you need anything else please let me know. Thanks, Ian.

Json below:

Response from curl -v -H "Accept: application/json" https://localhost:4443/debug/devices/ -k | python -m json.tool

is 

[
    {
        "@context": "https://iot.mozilla.org/schemas",
        "@type": [
            "SmartPlug"
        ],
        "actions": {},
        "adapterId": "thing-url-adapter",
        "description": "",
        "events": {},
        "id": "http---adapter.local-things-Relay",
        "name": "IoT-Bus Relay",
        "pin": {
            "pattern": null,
            "required": false
        },
        "pluginId": "thing-url-adapter",
        "properties": {
            "on": {
                "@type": "OnOffProperty",
                "name": "on",
                "type": "boolean",
                "value": false,
                "visible": true
            }
        },
        "uiHref": null
    },
    {
        "@context": "https://iot.mozilla.org/schemas",
        "@type": [
            "binarySensor"
        ],
        "actions": {},
        "adapterId": "thing-url-adapter",
        "description": "",
        "events": {},
        "id": "http---adapter.local-things-Touch",
        "name": "ESP32 Touch Input",
        "pin": {
            "pattern": null,
            "required": false
        },
        "pluginId": "thing-url-adapter",
        "properties": {
            "true": {
                "@type": "BooleanProperty",
                "name": "true",
                "type": "boolean",
                "value": false,
                "visible": true
            }
        },
        "uiHref": null
    },
    {
        "@context": "https://iot.mozilla.org/schemas",
        "@type": [
            "MotionSensor"
        ],
        "actions": {},
        "adapterId": "thing-url-adapter",
        "description": "",
        "events": {},
        "id": "http---adapter.local-things-HC-SR501",
        "name": "HC-SR501 PIR Sensor",
        "pin": {
            "pattern": null,
            "required": false
        },
        "pluginId": "thing-url-adapter",
        "properties": {
            "on": {
                "@type": "MotionProperty",
                "name": "on",
                "type": "boolean",
                "value": true,
                "visible": true
            }
        },
        "uiHref": null
    }
]

Each has a unique MAC address and IP address:

[image]

The output performing a direct GET on the IP addresses is as below:

[image]

mrstegeman, 2019-12-30:
This is over a year old. Please re-open if there is still an issue, since there have been quite a few changes since then.

closed 2019-12-30 by mrstegeman

lazylazyllama, 2020-04-11:
I'm using the current version of webthings-arduino and am facing the same issue ...