WebThingsIO/webthing-arduino #36: Async behavior

lgLindstrom, 2018-11-01:
I have studied the led example. Although is uses a async webserver you still need to "POLL" propertys to detect changes. 
Should it not be better to provide possibilities for a callback when a property is changed ?

hobinjk, 2018-11-01:
Callback registration would definitely be a nice thing to have for the library. The main question I'd have is whether it has a significant memory impact or could easily be implemented in a way that it could be selectively disabled on platforms where it wouldn't have benefits (i.e. the WiFi101WebThingAdapter for arduino boards)

lgLindstrom, 2018-11-02:
Will this work ( not sure about correct syntax, still struggling with C++
.. )

class BoolThingProperty : ThingProperty {
     public
BoolThingProperty

}

lgLindstrom, 2018-11-17:
Hi
I've looked a little bit on your code and noticed that "handleThingPropertyGet" in ESPWebThingAdapter.h is called repeatedly.
This is a bit confusing because I cant see any HTTP access to my device. Can you comment on this ?

hobinjk, 2018-11-26:
`handleThingPropertyGet` should be called repeatedly in response to thing-url-adapter's polling. It definitely shouldn't be called if there's no HTTP access so I'm not sure what could be going on there.

lgLindstrom, 2019-03-09:
Hmm,,, I added asynchronous behavior to my local clone,, how do I upload it ???

lgLindstrom, 2019-04-21:
Hi
I added a PR for my changes but it seems that I not is accepted. This is my first PR ever so I am not sure why. Can you direct me to some good sources of information to understand why it is rejected. I have not understood "anything"  from the sources I found.

gigs2go, 2019-08-13:
I come from Java API (not C++).
Property has a callback.
In this implementation, adding onSetProperty method pointer should be straightforward - then, when 'setValue' is called on ThingProperty, if method pointer not nulll, call it with ThingPropertyValue.
This would allow my code to specify behaviour when property is changed from 'outside'.

PR #55 (2020-01-07): Enhancement/async property
I have added, what I call Asynch Property functionallity. Not being able to test anything else than the ESPWebThingAdapter.
Also added a example.

Fixes #36
closed 2020-01-07 by mrstegeman