WebThingsIO/webthing-arduino #40: Allow reading the properties from gateway before setting them

jambuu, 2018-12-18:
As a user, I want to be able to retrieve properties from gateway before setting them.

Currently, the properties set via web interface are lost in case of a reset, turn off.

For example, if I need to set a property the color:

```c++
String color_ = "#DDDDDD"; //!< allocated for current value`
adapter = new WebThingAdapter("w25", ip_addr);

color_property_.setValue({.string = &color_});
nightlight_.addProperty(&color_property_);
```
When turning the arduino off, the last state is lost (although the gateway still has it). I understand that for this to work, there should be a support on the gateway side to query the last state of the device before it went off.

madb1lly, 2019-01-14:
I also want to do this and plan to try the solution to this closed issue: https://github.com/mozilla-iot/webthing-arduino/issues/30

mrstegeman, 2019-12-26:
I agree with @hobinjk #30 that this should just be stored in EEPROM. The web thing should be able to stand on its own without a gateway.

closed 2019-12-26 by mrstegeman