WebThingsIO/webthing-arduino #46: How to define a device and properties without using schemas

madb1lly, 2019-01-23:
The examples all seem to use schemas to define the devices and properties. How can devices and properties be defined without using the schemas, e.g. to define something for which a schema does not exist? Does the device or property have to be defined as a structure inside the Arduino code?

madb1lly, 2019-02-14:
Am I overcomplicating this? Is it just a case of defining a structure for in the Arduino code? Or is it necessary to define it in a JSON?

benfrancis, 2019-02-15:
I think you discussed this on IRC, but I will reply here for anyone following along.

Defining properties using this library will result in the device's web server exposing a Thing Description for the device which describes those properties in a JSON format which conforms with the [Thing Description](https://iot.mozilla.org/wot/#web-thing-description) specification.

You can define whatever properties, actions and events you like by specifying them in terms of their primitive data types, they don't have to conform to an existing schema. The Things Gateway will generate a user interface based on the Thing Description, using those primitive data types.

[Schemas](https://iot.mozilla.org/schemas/) define device capabilities in terms of common collections of properties, actions and events. If you tell the device to use one of these pre-defined schemas then additional semantic annotations will be added to the Thing Description which the Things Gateway (or any other WoT client) will use to generate a more sophisticated user interface for the device.

closed 2019-12-20 by mrstegeman