WebThingsIO/webthing-arduino #51: Not found esp

energo5, 2019-02-21:
Greetings.
I have an "gateway" installed on "orange pi zero" according to this instruction https://github.com/mozilla-iot/gateway/blob/master/README.md

Next, I installed on the "esp8266" webthing-arduino according to this instruction https://github.com/mozilla-iot/webthing-arduino

The web interface normally connects to orange pi (screen attached).
[image]

Http-request for esp is also normally processed (screen attached).
[image]

However, when searching for a new device, nothing is found (I attach a screen from the terminal connected to orange pi).
[image]

What could be the reason?

hobinjk, 2019-02-27:
It's possible that your local network doesn't support mDNS-based discovery. If you use the 'Add Thing by URL' option in the gateway you can manually specify "http://192.168.1.54/things/lamp" which should load properly and work even if the network doesn't want to do discovery.

If that doesn't work, I'd confirm that the gateway and the lamp can talk to each other by running `curl http://192.168.1.54/things/lamp` on the pi.

mrstegeman, 2019-02-27:
While doing some testing earlier, my Macbook was unable to see my ESP8266 via mDNS (but the other webthing libraries work fine), so I had to manually input the URL. I didn't dig in to figure out what was going on, though.

lgLindstrom, 2019-03-05:
Please, do we have a status update on this issue ?

mrstegeman, 2019-12-30:
I no longer see issues on either ESP32 or ESP8266. Closing this. Please re-open if there are further issues, but please remember that you can always add your things by URL in the gateway.

closed 2019-12-30 by mrstegeman

lazylazyllama, 2020-08-10:
I think this is still an issue.

I am writing a small firmware that uses web things and is compatible with esp32 and esp8266.
Here is my code: https://github.com/lazylazyllama/Incendio

So if I run it on a esp32, mdns is working and the gateway detects my device automatically.

But if I run it on a esp8266 it does not work. I can add the device by url and that works, but mdns does not work, which isn't ideal at all.

I even tried the code from here: https://hacks.mozilla.org/2018/04/making-a-web-thing-on-the-esp8266/
and it didn't work either.