WebThingsIO/webthing-arduino #52: gateway doesn't show up door sensor icon

ramachandrann, 2019-02-27:
Hello team,

Can you pls take a look at this issue I have with getting a door icon visible on things gateway 0.7 with esp8266

What am I missing ?
https://github.com/mozilla-iot/gateway/issues/1679

mrstegeman, 2019-02-27:
The code below works fine for me, with the following libraries (you may need to update):
* webthing-arduino - 0.4.1
* ArduinoJson - 5.13.4
* [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer/) - latest git version
* [ESPAsyncTCP](https://github.com/me-no-dev/ESPAsyncTCP) - latest git version

```cpp
#include <Thing.h>
#include <WebThingAdapter.h>

const char* ssid = "xxxx";
const char* password = "xxxx";

#define MAGNETIC_REED_PIN 2 //pin gpio 12 in sensor

WebThingAdapter *adapter;

const char* deviceTypes[] = {"DoorSensor", nullptr};
ThingDevice door("door", "Garage Door", deviceTypes);
ThingProperty doorOpen("open", "Whether the door is open", BOOLEAN, "OpenProperty");
ThingPropertyValue openValue;

void setup(void){
  pinMode(MAGNETIC_REED_PIN, INPUT);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  adapter = new WebThingAdapter("w25", WiFi.localIP());
  door.addProperty(&doorOpen);
  adapter->addDevice(&door);
  adapter->begin();
  
  Serial.println("HTTP server started");
  Serial.print("http://");
  Serial.print(WiFi.localIP());
  Serial.print("/things/");
  Serial.println(door.id);
}

void loop(void){
  adapter->update();

  if (digitalRead(MAGNETIC_REED_PIN)==HIGH) {
    openValue.boolean = true;
    Serial.println("Your Door is Open");
  }
  else {
    openValue.boolean = false;
    Serial.println("Your Door is Closed");
  }

  doorOpen.setValue(openValue);
}
```

I clicked the "Add by URL..." link on the gateway and used: `http://<ip-address>/things/door`

ramachandrann, 2019-02-27:
Thank you ... never realized this could have been the prob :) .. will take a look today

ramachandrann, 2019-02-27:
Perfect!!!!! Works like a charm. Now I'm going to put to use my orphaned esp8266s. Next is Alexa integration :)

closed 2019-02-27 by ramachandrann