WebThingsIO/webthing-arduino #53: missing mdns response.

lgLindstrom, 2019-03-03:
Hi
Running Esp32 with the LED example.   ( https://github.com/mozilla-iot/webthing-arduino/blob/master/examples/PlatformIO/LED/src/LED.cpp )

Seems like I cant get any answers from the device. My adapter looks like this..

```
   pairNow (){
    console.debug(' execution pairNow');   
    const browser = mdns.createBrowser(mdns.tcp('http'));
      browser.on('ready', () => {
        console.debug('Starting discovery...');
        browser.discover();
      });
      browser.on('update', (data) => {
        // The firmware from the http-on-off-wifi101 repository will wind up
        // creating a data record which looks like the following:
        //
        //  data: { addresses: [ '192.168.1.77' ],
        //  query: [],
        //  port: 80,
        //  fullname: 'http-on-off._moziot._tcp.local',
        //  txt: [ '_services' ],
        //  type:
        //   [ { name: 'dns-sd',
        //       protocol: 'udp',
        //       subtypes: [],
        //       description: undefined },
        //     { name: 'moziot',
        //       protocol: 'tcp',
        //       subtypes: [],
        //       description: undefined } ],
        //  host: 'wifi101-F714A9.local',
        //  interfaceIndex: 1,
        //  networkInterface: 'pseudo multicast' }
        console.debug('Recived data');
        console.debug(data);
        if (data.hasOwnProperty('fullname') && data.hasOwnProperty('host')) {
          const fullname = data.fullname;
          const host = data.host;
          if (typeof fullname === 'string' && typeof host === 'string') {
            if (fullname.startsWith('http-on-off.') &&
                host.startsWith('wifi101')) {
              let address = '';
              if (Array.isArray(data.addresses) && data.addresses.length > 0) {
                address = ` @ ${data.addresses[0]}`;
              }
              const url = `http://${host}`;
              console.log(`Adding ${url} (via mDNS Discovery ${address})`);
              this.handleDeviceAdded(
                new HttpOnOffDevice(this, 'HttpOnOffDevice', url));
            }
          }
        }
      });
  }

```

I am aware that the code maybe not is optional,, but at least it should generate a answer.

mrstegeman, 2019-03-04:
Duplicate of #51

hobinjk, 2019-03-05:
I think there's a fundamental misconception here. Any device running webthing-arduino should automatically be recognized by the gateway using the built-in [thing-url-adapter](https://github.com/mozilla-iot/thing-url-adapter). This means that if you click the "+" icon on the gateway's main page the device running webthing-arduino should show up automatically. If not, it's possible that mDNS isn't supported by your local network and you can instead manually enter the IP address of the device into the "Add Thing by URL" dialog available in the list of new things (e.g. http://192.168.1.54/things/lamp).

If you want to continue writing your own adapter the code you have above is specifying the wrong mDNS service type 'http' and would need to be `mdns.createBrowser(mdns.tcp('webthing'))` instead. However, I would recommend basing the code off of [thing-url-adapter's discovery implementation](https://github.com/mozilla-iot/thing-url-adapter/blob/master/thing-url-adapter.js#L670) since it's what the gateway uses and has a lot more testing than the old http-on-off-adapter code which is only compatible with devices running [this code](https://github.com/mozilla-iot/http-on-off-wifi101/blob/master/src/main.cpp)

lgLindstrom, 2019-03-06:
ahhh thank you. 
For some reason I was lead to believe that all examples etc was working with each other. I am gonna test this out.

I have another question about webthing-arduino. What about actions and events? I can see examples using properties but no one using actions and/or events.

PS, if documentation was improved, maybe you will avoid stupid questions from people like me.... DS

closed 2019-12-26 by mrstegeman