WebThingsIO/webthing-arduino #58: Missing `MDNS.update()` in `update()`

willdurand, 2019-03-17:
Hi,

using the latest Arduino 2.5.0 library and any example from this library (e.g., the LED sketch), mDNS does not work anymore: no device can be reached via its `<hostname>.local` domain.

According to [this comment](https://github.com/esp8266/Arduino/issues/5487#issuecomment-446642012), there should be a call to `MDNS.update()` in the `loop()` so this call should be added to https://github.com/mozilla-iot/webthing-arduino/blob/305860fdd3aba08914c4f5d2d82d17eadc9617e5/ESPWebThingAdapter.h#L74.

If I add `MDNS.update()` in the `loop()` of the LED sketch, the device is reachable again (via `ping w25.local`). The LWIP version does not matter (I tried v2 and v1.4).

hobinjk, 2019-03-18:
Thanks for bringing this up! I was wondering why we hadn't had an issue previously and reading through that issue was fascinating

closed 2019-03-18 by hobinjk

willdurand, 2019-03-18:
❤️

commit 5cc28d91 (2019-03-18): Add missing call to MDNS.update for change in library behavior

Fix #58