WebThingsIO/webthing-arduino #59: Exception when trying to `GET /`

willdurand, 2019-03-17:
Hi,

using the LED sketch, browsing `/things/led` from the command line works well but trying to get `/` crashes the device (Wemos D1 mini) with the following exception:

```
Connected to wuerstchen
IP address: 192.168.2.92
MDNS responder started
HTTP server started
http://192.168.2.92/things/led

Exception (28):
epc1=0x4021007d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000028 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffdb20 end: 3fffffb0 offset: 01a0
3fffdcc0:  00000020 00000000 3fff12d4 4020778d  
3fffdcd0:  3fffdd30 00000000 3fff12d4 402077c8  
3fffdce0:  3fff0e5c 402186b0 00000000 40100620  
3fffdcf0:  00000000 3fff04a4 00000078 3fff04a4  
[... 400 more lines]
```

willdurand, 2019-03-17:
The reason seems to be the JSON buffer being assigned to `4096`. I changed to `2048` and I cannot reproduce the bug in the OP. According to the ArduinoJson docs:

>  As a general rule, if your StaticJsonBuffer is bigger than 2KB, then it may be a good time to switch to a DynamicJsonBuffer.

Should we change that in `handleThings`?

hobinjk, 2019-03-18:
I recently landed a patch changing all the StaticJsonBuffers to DynamicJsonBuffers in WiFi101WebThingAdapter. I agree that removing StaticJsonBuffers is a good idea since all of them are gigantic. I'll remove the remaining ones in EthernetWebThingAdapter.

willdurand, 2019-03-18:
Reading your reply made me realize that I did not specify any context. The issue happens with an ESP board and the JSON buffer is actually dynamic already, yet it "reserves" (I suppose) 4KB:

https://github.com/mozilla-iot/webthing-arduino/blob/305860fdd3aba08914c4f5d2d82d17eadc9617e5/ESPWebThingAdapter.h#L128

hobinjk, 2019-03-18:
Ah, that makes more sense! I'll knock down that size accordingly. The buffer size reservations are based on the previous StaticJsonBuffer sizes which in turn were determined when I was running a bunch of complex things off one ESP8266 so they're less than representative of sensible defaults.

closed 2019-03-18 by hobinjk

willdurand, 2019-03-18:
Thanks for fixing all these minor issues so fast! Could you please release a new version?

commit ee99c613 (2019-03-18): Remove the remaining StaticJsonBuffers

Fix #59