WebThingsIO/webthing-arduino #85: Dynamicly add Wifi credential insted of hard coding

Ramjivan, 2019-08-19:
Hi folks,

I was to add WiFiManager Library to provide feature of adding WiFi credentials dynamically,
Later I found it was using ESP8266Webserver Library and the same lib is being utilized by webthing lib.  

Can anybody tell me how can I resolve this issue.

Here's the error I'm encountering 

    In file included from C:\Users\RAM\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.h:17:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:4:

C:\Users\RAM\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:19: error: redeclaration of 'HTTP_ANY'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                   ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\webthing-arduino/Thing.h:16:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:2:

C:\Users\RAM\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:63:3: note: previous declaration 'WebRequestMethod HTTP_ANY'

   HTTP_ANY     = 0b01111111,

   ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.h:17:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:4:

C:\Users\RAM\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:29: error: redeclaration of 'HTTP_GET'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                             ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\webthing-arduino/Thing.h:16:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:2:

C:\Users\RAM\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:56:3: note: previous declaration 'WebRequestMethod HTTP_GET'

   HTTP_GET     = 0b00000001,

   ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.h:17:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:4:

C:\Users\RAM\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:39: error: redeclaration of 'HTTP_POST'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                       ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\webthing-arduino/Thing.h:16:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:2:

C:\Users\RAM\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:57:3: note: previous declaration 'WebRequestMethod HTTP_POST'

   HTTP_POST    = 0b00000010,

   ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.h:17:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:4:

C:\Users\RAM\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:50: error: redeclaration of 'HTTP_PUT'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                                  ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\webthing-arduino/Thing.h:16:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:2:

C:\Users\RAM\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:59:3: note: previous declaration 'WebRequestMethod HTTP_PUT'

   HTTP_PUT     = 0b00001000,

   ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.h:17:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:4:

C:\Users\RAM\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:60: error: redeclaration of 'HTTP_PATCH'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                                            ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\webthing-arduino/Thing.h:16:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:2:

C:\Users\RAM\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:60:3: note: previous declaration 'WebRequestMethod HTTP_PATCH'

   HTTP_PATCH   = 0b00010000,

   ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.h:17:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:4:

C:\Users\RAM\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:72: error: redeclaration of 'HTTP_DELETE'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                                                        ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\webthing-arduino/Thing.h:16:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:2:

C:\Users\RAM\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:58:3: note: previous declaration 'WebRequestMethod HTTP_DELETE'

   HTTP_DELETE  = 0b00000100,

   ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.h:17:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:4:

C:\Users\RAM\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:85: error: redeclaration of 'HTTP_OPTIONS'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                                                                     ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\webthing-arduino/Thing.h:16:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:2:

C:\Users\RAM\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:62:3: note: previous declaration 'WebRequestMethod HTTP_OPTIONS'

   HTTP_OPTIONS = 0b01000000,

   ^

In file included from C:\Users\RAM\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,

                 from C:\Users\RAM\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,

                 from C:\Users\RAM\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Arduino.h:32,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:1:

C:\Users\RAM\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol

   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))

                                                                                                                                  ^

C:\Users\RAM\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.h:26:24: note: in expansion of macro 'PROGMEM'

 const char HTTP_HEAD[] PROGMEM = "<!DOCTYPE html><html lang=\"en\"><head><meta charset=\"UTF-8\" name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";

                        ^

In file included from C:\Users\RAM\Documents\Arduino\libraries\webthing-arduino/Thing.h:16:0,

                 from C:\Users\RAM\Desktop\xpanar1\webthing\webthing.ino:2:

C:\Users\RAM\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:61:3: error: previous declaration of 'WebRequestMethod HTTP_HEAD'

   HTTP_HEAD    = 0b00100000,

   ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

gigs2go, 2019-08-21:
Quick look shows that WiFiManager is using ESP8266WebServer, whereas ESPWebThingAdapter uses ESP8266AsynWebServer. Both define the HTTP methods. It might be possible to combine them (using #define and #ifdef) ... or mod'ing the WiFiManager to use Async might do it.
ESPWebThingAdapter should definitely (IMO) use the Async version.

Ramjivan, 2019-08-24:
> Quick look shows that WiFiManager is using ESP8266WebServer, whereas ESPWebThingAdapter uses ESP8266AsynWebServer. Both define the HTTP methods. It might be possible to combine them (using #define [...]

Modifying WiFiManager to use Async does not solves the problem

mrstegeman, 2019-12-30:
This seems like an issue with the web server libraries, not directly with this library. Closing.

closed 2019-12-30 by mrstegeman

fam4r, 2020-05-07:
@Ramjivan you tried with https://github.com/alanswx/ESPAsyncWiFiManager ?