WebThingsIO/webthing-arduino #86: WebServer configuration

gigs2go, 2019-08-21:
Specifically for ESPAsyncWebServer ...
I would like to be able to configure my webthing device as an AP/captive portal, serving a page from SPIFFS. The ESPWebThingAdapter creates (and hides) the server itself.
I have modified the code in the ESPWebThingAdapter to do this, but it's a very specific use-case.
I would like to create the server before creating the WebThingAdapter, and pass it as a construction parameter (or in an init method) so that the Adapter can be seen as an 'overlay' to the existing server - which has been pre-configured to my needs.

Excellent project! I am following with interest, and am looking forward to actions and events.
Many Thanks

Lyntball, 2019-11-14:
I second this

mrstegeman, 2019-12-30:
This is probably outside the scope of this library. However, another user on Discourse has been working on this very problem, by combining webthing-arduino and IotWebConf. See here for the discussion: https://discourse.mozilla.org/t/arduino-libs-or-tasmota/51142

closed 2019-12-30 by mrstegeman