WebThingsIO/webthing-arduino #96: Premature termination JSON output

qistoph, 2019-11-09:
Working with an ESP8266 (NodeMCU v3) I'm having issues with JSON output being truncated. Resulting in invalid JSON data.

I have 11 properties;
- Internal LED
- Motion sensor
- Temperature sensor
- 8 ports on a relays board (with an MCP23008 controller)

Surfing to `http://MY-IP/` my browser shows the error:
**SyntaxError: JSON.parse: unterminated string at line 1 column 3612 of the JSON data**

Not adding either the LED or motion sensor property makes the sketch return valid JSON data.
https://gist.github.com/qistoph/bfdf8601969078be9284bcf6224351bf#file-thingmcp23008-ino-L98

Below are the number of characters returned, determined with `curl 'http://MY-IP/' | wc -c`:
- Neither disabled: `3860` (**invalid JSON because of termination**)
- LED disabled: `4817`
- Motion disabled: `4796`

If anyone has any suggestions to fix or further troubleshoot the issue, I'd love to hear from you.

hobinjk, 2019-11-09:
This sounds like ArduinoJson might not be able to size up the buffer past a certain point. It's possible that my migration to ArduinoJson 6 in https://github.com/mozilla-iot/webthing-arduino/pull/95 will fix this issue but testing that out would require manually checking out that branch and making sure you upgrade your local ArduinoJson.

As a workaround, you could move some of the properties to a second thing and add both to the adapter with multiple calls to `adapter->addDevice(&thingOne);`

qistoph, 2019-11-11:
Thanks @hobinjk for your suggestions. I have checked out your branch and compiled my project with ArduinoJson 6.13.0.

The generated JSON was no longer broken, but only showed the first two Things in the JSON generated.

The work-around didn't work for me, unfortunately. I was actually already adding Things/Devices instead of properties. I'll rephrase my initial issue accordingly.

mijapa, 2020-01-01:
Working with an ESP32.
After migration from 0.4.1 to 0.10.1 version of library I'm having issues with JSON output being truncated.

mrstegeman, 2020-01-01:
You may want to try changing the size of some of the JSON buffers. For instance:
* https://github.com/mozilla-iot/webthing-arduino/blob/a67d230f2f54c29213a2ee27176a3f054d94b922/ESPWebThingAdapter.h#L274
* https://github.com/mozilla-iot/webthing-arduino/blob/a67d230f2f54c29213a2ee27176a3f054d94b922/ESPWebThingAdapter.h#L436

They're set fairly small by default.

mijapa, 2020-01-02:
Thank you @mrstegeman 
Changing buff size in line 274 to 4096 helped.

mrstegeman, 2020-01-02:
The buffer sizes are now configurable as such:
```cpp
// By default, buffers are 256 bytes for small documents, 1024 for larger ones

// To use a pre-defined set of larger JSON buffers (4x larger)
#define LARGE_JSON_BUFFERS 1

// Else, you can define your own size
#define SMALL_JSON_DOCUMENT_SIZE <something>
#define LARGE_JSON_DOCUMENT_SIZE <something>
```

commit f00574f3 (2020-01-02): Make JSON document sizes configurable.

Fixes #96
closed 2020-01-02 by mrstegeman