WebThingsIO/webthing-arduino #107: Exception error crash during update

DSiwi, 2020-04-09:
Hi.
I have reasonably basic application for a system consisting of a weather sensors (BME680), a soil moisture sensor, and some housekeeping monitoring like supply and battery voltage (this being intended as a solar panel powered device.
It has never run, because when adapter->update is called it crashes with Exception error 28.
In the end I removed all webthing code from my project, ensured it ran OK as stand-alone, and then inserted the webthing code from the BME280 example in this repository. I made only the minimum changes to use my variable names. That also crashes with the same error.
I used  "me-no-dev/EspExceptionDecoder" to break down the stack information (with DEBUG set to "Core", and got this:

Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x402030f4: loop() at C:\Users\Derek\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h line 182
EXCVADDR: 0x00000070

Decoding stack results
0x4020a7d4: loop_wrapper() at C:\Users\Derek\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 180

I am using Arduino v1.8.12 in Win7. All libraries are up to date.
Any help would be great.
Thanks, Derek

mrstegeman, 2020-04-09:
Can you share your code?

DSiwi, 2020-04-09:
Thanks for getting back so back.
Code is attached (with a couple of comments on this) and a couple of other odds and ends that might be useful. I have tried a few things like different Flash sizes, but no success.
[Soil_Sensor_IoTtest.zip](https://github.com/mozilla-iot/webthing-arduino/files/4457859/Soil_Sensor_IoTtest.zip)

mrstegeman, 2020-04-09:
The problem is that you forgot to call `setupWebThing()` in your `setup()` function. All seems good with that added in.

closed 2020-04-09 by mrstegeman