WebThingsIO/webthing-arduino #109: Still facing JSON truncation issues despite LARGE_JSON_BUFFERS

lstwn, 2020-04-13:
Hey,

this is related to #96. I am using an ESP 8266 device and I'm trying to get a simple temperature sensor working as a web thing.

I tried two setups due to the hint mentioned in #96:
1. one device with three properties (location, temperature, humidity) [code](https://gist.github.com/lstwn/5b9a42b9d546aa29f8231751060de0fb)
2. two devices [code](https://gist.github.com/lstwn/385e2d8314201a754c159cf1e1b2b186)
    * one with location and temperature
    * one with location and humidity

Both do not work. 

The first setup suffers from the truncation issues despite setting `#define LARGE_JSON_BUFFERS 1` or `#define SMALL_JSON_DOCUMENT_SIZE <something>` and
`#define LARGE_JSON_DOCUMENT_SIZE <something>` with different values, but no luck. The properties itself work because when navigating to `https://[...]/properties` they show up with the right values from the sensor. But the device view is truncated.

The second setup is even more weird. Could also be an error in my code though. Both devices show the location as specified, but also _only_ the humidity and no temperature although I assigned them correctly I think. This is also true for the `https://[...]/properties` path: Even on the temperature device there is only location and (falsely) humidity shown despite not calling `addProperty` on that device..

Let me know if I can help with anything else.

lazylazyllama, 2020-04-13:
I think `#define LARGE_JSON_BUFFERS 1` has to set before all your includes... otherwise those libs will ignore it.

mrstegeman, 2020-04-13:
> I think `#define LARGE_JSON_BUFFERS 1` has to set before all your includes... otherwise those libs will ignore it.

Yes, this is exactly right.

closed 2020-04-13 by mrstegeman