WebThingsIO/webthing-arduino #110: changedValueOrNull() is slow or something?

lazylazyllama, 2020-04-13:
I have multiple properties and I realized, that when I call multiple `deviceX.changedValueOrNull()` methods inside the loop function to check if any changes happened, my whole program is behaving very weird.
If I only call the first one its still working okay I think... but the more of those methods I call the worse it gets I think.. My device is getting offline in the gateway and changes take a lot of time to get done sometimes.. also my device gets into weird states where the gateway think its off but its on and so on...

Do you know why this is happening?
I really would like to use those methods in the loop function as they make things so much cleaner and I don't have to save old states somewhere to detect changes...

Code:
```c++
void loop(void) {
  if (deviceOn.changedValueOrNull() != nullptr) {
    Serial.println(on);
  }

  if (deviceBrightness.changedValueOrNull() != nullptr) {
    Serial.println(brightness);
  }
}
```

mrstegeman, 2020-04-13:
You're missing a call to `adapter->update()`. Try adding that back in and see if things look better.

lazylazyllama, 2020-04-13:
Oh sorry, but I do coll it. I just stripped my code a bit down to make it less confusing

lazylazyllama, 2020-04-13:
This would be all:

```c++
void update(String *color, float const brightness) {
  if (!color) {
    return;
  }
  float dim = brightness / 100.0;
  int red, green, blue;
  if (color && (color->length() == 7) && color->charAt(0) == '#') {
    const char *hex = 1 + (color->c_str()); // skip leading '#'
    sscanf(0 + hex, "%2x", &red);
    sscanf(2 + hex, "%2x", &green);
    sscanf(4 + hex, "%2x", &blue);
  }
  #ifdef ESP32
    analogWrite(redPin, red * dim, 255U);
    analogWrite(greenPin, green * dim, 255U);
    analogWrite(bluePin, blue * dim, 255U);
  #elif defined(ESP8266)
    analogWrite(redPin, red * dim);
    analogWrite(greenPin, green * dim);
    analogWrite(bluePin, blue * dim);
  #endif
}

void loop(void) {
  if (WiFi.status() != WL_CONNECTED) {
    WifiManager::begin();
  }
  OTAManager::handle();

  adapter->update();

  bool on = deviceOn.getValue().boolean;
  float brightness = deviceBrightness.getValue().number;
  update(&color, on ? brightness : 0);

  digitalWrite(LED_BUILTIN, on ? HIGH : LOW);
  if (deviceOn.changedValueOrNull() != nullptr) {
    Serial.println(on);
  }

  if (deviceBrightness.changedValueOrNull() != nullptr) {
    Serial.println(brightness);
  }
}
```

mrstegeman, 2020-04-13:
`changedValueOrNull` is really just meant to be called internally, as that is what the adapter uses to figure out which properties to send updates for via the WebSocket. It was never meant to be used by end users. What you're probably doing is causing races between your code and the library, as that function actually modifies internal state.

Instead, you can save off a value and check that in your `loop()` function. There's a basic example of this in the LED example: https://github.com/mozilla-iot/webthing-arduino/blob/master/examples/LED/LED.ino#L75-L85

lazylazyllama, 2020-04-13:
Okay yeah I did it like this before, but hoped to make my code more clean by using those function calls

closed 2020-04-13 by mrstegeman