WebThingsIO/webthing-arduino #112: Gateway does not notice if color or colormode are changed

lazylazyllama, 2020-04-13:
I am using webthing-arduino and I just copied this example: https://github.com/mozilla-iot/webthing-arduino/blob/master/examples/RGBLamp/RGBLamp.ino

What I did is: I changed the color in the Webthings Gateway.

Result:
If I fetch the json over the endpoint of my `Thing` I get the new color as a result. So my `Thing` got the color and that part works fine.

Problem:
But The color of the LED Icon on the Webthings Gateway does not change. If I reload the Page also the colorpicker will have its initial value.. 
I don't only have this issue with the color picker, but also if I change things like the colorMode from within the `Thing`.. the JSON is always correct, but the Gateway does not recognize the change...

Do I have to send a message from the `Thing` to the Gateway so that the Gateway will now, that properties have changed? Or why isn't this working?

mrstegeman, 2020-04-13:
Fix is in v0.11.3.

closed 2020-04-13 by mrstegeman

lazylazyllama, 2020-04-13:
Thank you so much :)

PR #111 (2020-04-13): Fix string property change notifications.
Fixes #112