WebThingsIO/webthing-arduino #114: ArduinoJson does not compile

Lux998, 2020-06-04:
When i try to compile an example it returns rhis error :

Arduino:1.8.12 (Linux), Scheda:"Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantContent.hpp:11:0,
                 from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantSlot.hpp:9,
                 from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/MemoryPool.hpp:11,
                 from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Strings/RamStringAdapter.hpp:7,
                 from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Strings/StringAdapters.hpp:8,
                 from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Misc/SerializedValue.hpp:7,
                 from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:7,
                 from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/SlotFunctions.hpp:8,
                 from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:7,
                 from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:17,
                 from /home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /home/lux/Arduino/libraries/webthing-arduino/Thing.h:22,
                 from /home/lux/Arduino/libraries/webthing-arduino/examples/LED/LED.ino:13:
/home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp: In instantiation of 'bool ArduinoJson6152_0000010::variantSetInteger(ArduinoJson6152_0000010::VariantData*, T) [with T = long long int]':
/home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:187:45:   required from 'bool ArduinoJson6152_0000010::VariantRef::set(T, typename ArduinoJson6152_0000010::enable_if<ArduinoJson6152_0000010::is_integral<T>::value>::type*) const [with T = long long int; typename ArduinoJson6152_0000010::enable_if<ArduinoJson6152_0000010::is_integral<T>::value>::type = void]'
/home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:44:5:   required from 'typename ArduinoJson6152_0000010::enable_if<(! ArduinoJson6152_0000010::is_array<TValue>::value), ArduinoJson6152_0000010::MemberProxy<TParent, TStringRef>&>::type ArduinoJson6152_0000010::MemberProxy<TParent, TStringRef>::operator=(const TValue&) [with TValue = long long int; TObject = ArduinoJson6152_0000010::ObjectRef; TStringRef = String; typename ArduinoJson6152_0000010::enable_if<(! ArduinoJson6152_0000010::is_array<TValue>::value), ArduinoJson6152_0000010::MemberProxy<TParent, TStringRef>&>::type = ArduinoJson6152_0000010::MemberProxy<ArduinoJson6152_0000010::ObjectRef, String>&]'
/home/lux/Arduino/libraries/webthing-arduino/Thing.h:316:22:   required from here
/home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp:135:3: error: static assertion failed: To use 64-bit integers with ArduinoJson, you must set ARDUINOJSON_USE_LONG_LONG to 1. See https://arduinojson.org/v6/api/config/use_long_long/
   ARDUINOJSON_ASSERT_INTEGER_TYPE_IS_SUPPORTED(T);
   ^
/home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp: In instantiation of 'typename ArduinoJson6152_0000010::enable_if<ArduinoJson6152_0000010::is_integral<T>::value, T>::type ArduinoJson6152_0000010::variantAs(const ArduinoJson6152_0000010::VariantData*) [with T = long long int; typename ArduinoJson6152_0000010::enable_if<ArduinoJson6152_0000010::is_integral<T>::value, T>::type = long long int]':
/home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp:93:27:   required from 'T ArduinoJson6152_0000010::variantAs(ArduinoJson6152_0000010::VariantData*, ArduinoJson6152_0000010::MemoryPool*) [with T = long long int]'
/home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:247:63:   required from 'typename ArduinoJson6152_0000010::VariantAs<T>::type ArduinoJson6152_0000010::VariantRef::as() const [with T = long long int; typename ArduinoJson6152_0000010::VariantAs<T>::type = long long int]'
/home/lux/Arduino/libraries/webthing-arduino/Thing.h:605:53:   required from here
/home/lux/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp:58:3: error: static assertion failed: To use 64-bit integers with ArduinoJson, you must set ARDUINOJSON_USE_LONG_LONG to 1. See https://arduinojson.org/v6/api/config/use_long_long/
   ARDUINOJSON_ASSERT_INTEGER_TYPE_IS_SUPPORTED(T);
   ^
exit status 1

If I add a 
`#//#define ARDUINOJSON_USE_LONG_LONG 1` it stops giving this error,
is this the right procedure?

mrstegeman, 2020-06-04:
I've never needed to use that, but if it solves your problem, then I don't see an issue with it.

closed 2020-06-04 by mrstegeman

madb1lly, 2020-06-05:
Hi @mrstegeman and @Lux998,

I needed to add this line when building in Ubuntu 1804 with VSCode. On Win10 with VSCode and Arduino IDE this line is not necessary.

I've not noticed any difference in the performance or behaviour of devices with that line in, so I think it's okay to add it regardless of the OS being used for building.

Cheers 🙂