WebThingsIO/webthing-arduino #116: ARDUINOJSON_USE_LONG_LONG must be set to 1

schnef, 2020-07-09:
I had to set ARDUINOJSON_USE_LONG_LONG to 1 before including Thing.h. See [ARDUINOJSON_USE_LONG_LONG](https://arduinojson.org/v6/api/config/use_long_long/). Using 64 bits instead of 32, is that an issue?

Lib versions used:
AdruinoJson 6.15.2
webthing-arduino 0.11.5

Board:
ITEAD sonoff basic

Compiler error message

```
/home/lemoi/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp: In instantiation of 'bool ArduinoJson6152_0000010::variantSetInteger(ArduinoJson6152_0000010::VariantData*, T) [with T = long long int]':
/home/lemoi/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:187:45:   required from 'bool ArduinoJson6152_0000010::VariantRef::set(T, typename ArduinoJson6152_0000010::enable_if<ArduinoJson6152_0000010::is_integral<T>::value>::type*) const [with T = long long int; typename ArduinoJson6152_0000010::enable_if<ArduinoJson6152_0000010::is_integral<T>::value>::type = void]'
/home/lemoi/Arduino/libraries/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:44:5:   required from 'typename ArduinoJson6152_0000010::enable_if<(! ArduinoJson6152_0000010::is_array<TValue>::value), ArduinoJson6152_0000010::MemberProxy<TParent, TStringRef>&>::type ArduinoJson6152_0000010::MemberProxy<TParent, TStringRef>::operator=(const TValue&) [with TValue = long long int; TObject = ArduinoJson6152_0000010::ObjectRef; TStringRef = String; typename ArduinoJson6152_0000010::enable_if<(! ArduinoJson6152_0000010::is_array<TValue>::value), ArduinoJson6152_0000010::MemberProxy<TParent, TStringRef>&>::type = ArduinoJson6152_0000010::MemberProxy<ArduinoJson6152_0000010::ObjectRef, String>&]'
/home/lemoi/Arduino/libraries/webthing-arduino/Thing.h:316:22:   required from here
/home/lemoi/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp:135:3: error: static assertion failed: To use 64-bit integers with ArduinoJson, you must set ARDUINOJSON_USE_LONG_LONG to 1. See https://arduinojson.org/v6/api/config/use_long_long/
   ARDUINOJSON_ASSERT_INTEGER_TYPE_IS_SUPPORTED(T);
   ^
```

madb1lly, 2020-07-09:
Hi @schnef,

I've also had to use this on Ubuntu with VS Code but not on Windows with Arduino IDE, so I'm not sure what makes this required. In any case I've noticed any difference in performance or behaviour.

Cheers 🙂

schnef, 2020-07-09:
I'm running on a Linux Debian 10 64 bit machine. That's the only link I see with long_long

mrstegeman, 2020-07-09:
I recently committed a fix for that, which should be in the next release.

https://github.com/mozilla-iot/webthing-arduino/blob/master/Thing.h#L22

closed 2020-07-09 by mrstegeman