WebThingsIO/webthing-arduino #118: Give the possibility to create custom endpoints

apiel, 2020-08-05:
Would it be possible to define some custom endpoint. I was trying something like:

```c
  adapter->server.on("/hello", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200, "text/plain", "Hello, world");
  });
```
but I am getting the following error: `error: 'AsyncWebServer WebThingAdapter::server' is private`

In my firmware, I am using an http server for multiple features, for example to to pair the device with Amazon Alexa or to trigger OTA updates. When I try to add webthing-arduino, it is conflicting. The best would be to be able to re-use the server instance from webthing-arduino.

mrstegeman, 2020-08-05:
We actually have this capability in all of our other webthing libraries. However, it's difficult to do so with this one, since essentially each platform uses a different web server library. It's not straightforward to write an agnostic interface for adding new routes. We _could_ make the server member public, I suppose, but that feels so unfinished.

apiel, 2020-08-05:
Right now the only way I found is to create 2 servers. One for the webthings adapter on port 8080 and one using ESPAsyncWebServer on port 80. Would be amazing to be able to have both on port 80. Of course making `server` public could bring inconsistency if users overwrite some endpoints defined by webthings, but it would be an easy way to start...