WebThingsIO/webthing-arduino #128: PushButton PressedEvent

miathedev, 2021-01-02:
I tryed to create a PushButton Device. Basically a remote:

```
#include <Arduino.h>

#include "wifisecret.h"
#define LARGE_JSON_BUFFERS 1

#include <Arduino.h>
#include <Thing.h>
#include <WebThingAdapter.h>

const uint8_t buttonPin = 4;

WebThingAdapter *adapter;
//https://webthings.io/schemas/#PushButton
const char *remoteTypes[] = {"PushButton", nullptr};
ThingDevice remote("urn:dev:ops:mypushbutton", "Push Button I", remoteTypes);

ThingEvent pressed("pressed",
                   "PressedEvent",
                   BOOLEAN, "PressedEvent");

ThingProperty button("button", "PushButton", BOOLEAN, "PushedProperty");

void setup(void)
{
  pinMode(buttonPin, INPUT);

  Serial.begin(115200);
  Serial.setDebugOutput(true);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  adapter = new WebThingAdapter("push-button", WiFi.localIP());

  remote.description = "A Webthings PushButton";

  button.title = "PushButton";

  remote.addProperty(&button);
  remote.addEvent(&pressed);
  adapter->addDevice(&remote);
  adapter->begin();

  Serial.println("HTTP server started");
  Serial.print("http://");
  Serial.print(WiFi.localIP());
  Serial.print("/things/");
  Serial.println(remote.id);

  randomSeed(analogRead(0));
}

unsigned long oldMillis = 0;
bool test = true;
void loop(void)
{
  if (millis() >= oldMillis + 5000)
  {
    Serial.println("FOrce Event!");
    oldMillis = millis();

    test = !test;
    button.setValue({.boolean = test});
    
    ThingEventObject *ev = new ThingEventObject("pressed", BOOLEAN, {.boolean = true});
    remote.queueEventObject(ev);
  }
  adapter->update();
}
```
What works:

The Node can be found by discovery. WebThings Gateway receives every 5 seconds a button change.

What does not work:

A PressedEvent should occur every 5 seconds. But it does not.

Does someone know what i made wrong?
With kind regards,
Mia

mrstegeman, 2021-01-02:
Hmm, the code _looks_ right. What kind of device are you running on?

miathedev, 2021-01-02:
The "Sketch" itself runs on a esp8266 - WeMos D1 mini.
As Gateway im using the current prebuilt raspberry pi OS image (gateway-0.12.0.img) - running on a raspberry pi 3.

mrstegeman, 2021-01-02:
Ok, I have an ESP8266 I can test with in a couple days.

However, I’d strongly recommend you update your gateway to 1.0.0 before we shut down all of our old Mozilla infrastructure on Monday.

mrstegeman, 2021-01-04:
The problem is that all of your events had the same default timestamp. Here is an example of how to fix that, using the [ESPDateTime](https://github.com/mcxiaoke/ESPDateTime) library:

```cpp
#include "wifisecret.h"
#define LARGE_JSON_BUFFERS 1

#include <Arduino.h>
#include <ESPDateTime.h>
#include <Thing.h>
#include <WebThingAdapter.h>

const uint8_t buttonPin = 4;

WebThingAdapter *adapter;
//https://webthings.io/schemas/#PushButton
const char *remoteTypes[] = {"PushButton", nullptr};
ThingDevice remote("urn:dev:ops:mypushbutton", "Push Button I", remoteTypes);

ThingEvent pressed("pressed",
                   "PressedEvent",
                   BOOLEAN, "PressedEvent");

ThingProperty button("button", "PushButton", BOOLEAN, "PushedProperty");

void setup(void)
{
  pinMode(buttonPin, INPUT);

  Serial.begin(115200);
  Serial.setDebugOutput(true);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  adapter = new WebThingAdapter("push-button", WiFi.localIP());

  remote.description = "A Webthings PushButton";

  button.title = "PushButton";

  remote.addProperty(&button);
  remote.addEvent(&pressed);
  adapter->addDevice(&remote);
  adapter->begin();

  Serial.println("HTTP server started");
  Serial.print("http://");
  Serial.print(WiFi.localIP());
  Serial.print("/things/");
  Serial.println(remote.id);

  randomSeed(analogRead(0));

  DateTime.begin();
  if (!DateTime.isTimeValid()) {
    Serial.println("Failed to get time from server.");
  }
}

unsigned long oldMillis = 0;
bool test = true;
void loop(void)
{
  if (millis() >= oldMillis + 5000) {
    if (!DateTime.isTimeValid()) {
      Serial.println("Failed to get time from server, retry.");
      DateTime.begin();
    } else {
      Serial.println("Force Event!");
      oldMillis = millis();
  
      test = !test;
      button.setValue({.boolean = test});
      
      ThingEventObject *ev = new ThingEventObject(
        "pressed",
        BOOLEAN,
        {.boolean = true},
        DateTime.formatUTC(DateFormatter::ISO8601)
      );
      remote.queueEventObject(ev);
  
      Serial.println();
    }
  }
  
  adapter->update();
}
```

closed 2021-01-06 by mrstegeman