WebThingsIO/webthing-arduino #130: Not compiling webthing-arduino 0.12.0

XauTrop, 2021-01-26:
trying to compile just:
#include <Arduino.h>
#include "Thing.h"
#include "WebThingAdapter.h"
I got these errors:
In file included from C:\Users\Jaume\Documents\Arduino\sketch_jan26b\sketch_jan26b.ino:13:0:
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:63:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
   DynamicJsonDocument *actionRequest = nullptr;
           ^
In file included from C:\Users\Jaume\Documents\Arduino\sketch_jan26b\sketch_jan26b.ino:13:0:
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:70:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
   ThingActionObject(const char *name_, DynamicJsonDocument *actionRequest_,
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:149:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
   ThingActionObject *(*generator_fn)(DynamicJsonDocument *);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:160:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
               ThingActionObject *(*generator_fn_)(DynamicJsonDocument *))
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:164:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
               ThingActionObject *(*generator_fn_)(DynamicJsonDocument *))
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:169:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
               ThingActionObject *(*generator_fn_)(DynamicJsonDocument *))
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:173:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
   ThingActionObject *create(DynamicJsonDocument *actionRequest) {
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:513:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     DynamicJsonDocument message(LARGE_JSON_DOCUMENT_SIZE);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:518:11: error: serializeJson is a function from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     serializeJson(message, jsonStr);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:618:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
   ThingActionObject *requestAction(DynamicJsonDocument *actionRequest) {
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:679:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     DynamicJsonDocument message(SMALL_JSON_DOCUMENT_SIZE);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:684:11: error: serializeJson is a function from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     serializeJson(message, jsonStr);
           ^
In file included from C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/WebThingAdapter.h:13:0,
                 from C:\Users\Jaume\Documents\Arduino\sketch_jan26b\sketch_jan26b.ino:14:
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:232:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
   void sendErrorMsg(DynamicJsonDocument &prop, AsyncWebSocketClient &client,
           ^
In file included from C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/WebThingAdapter.h:13:0,
                 from C:\Users\Jaume\Documents\Arduino\sketch_jan26b\sketch_jan26b.ino:14:
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:237:11: error: serializeJson is a function from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     serializeJson(prop, jsonStr);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:268:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     DynamicJsonDocument newProp(SMALL_JSON_DOCUMENT_SIZE);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:269:11: error: deserializeJson is a function from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     auto error = deserializeJson(newProp, rawData, len);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:290:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
         DynamicJsonDocument *actionRequest =
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:291:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
             new DynamicJsonDocument(SMALL_JSON_DOCUMENT_SIZE);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:321:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     DynamicJsonDocument message(LARGE_JSON_DOCUMENT_SIZE);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:336:11: error: serializeJson is a function from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
       serializeJson(message, jsonStr);
[... 76 lines cut ...]
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:636:11: error: serializeJson is a function from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     serializeJson(item, jsonStr);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:655:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     DynamicJsonDocument doc(LARGE_JSON_DOCUMENT_SIZE);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:658:11: error: serializeJson is a function from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     serializeJson(queue, *response);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:683:11: error: DynamicJsonDocument is a class from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     DynamicJsonDocument newBuffer(SMALL_JSON_DOCUMENT_SIZE);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:684:11: error: deserializeJson is a function from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     auto error = deserializeJson(newBuffer, body_data);
           ^
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/ESPWebThingAdapter.h:704:11: error: serializeJson is a function from ArduinoJson 6 but version 5 is installed. Visit arduinojson.org to get more information.
     serializeJson(newProp, *response);
           ^
Se encontraron varias bibliotecas para "WiFi.h"
Usado: C:\Users\Jaume\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 No usado: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compilando para la tarjeta AESP32 Dev Module

Any help?

mrstegeman, 2021-01-26:
.You need to update the version of ArduinoJson you have installed to at least version 6.17.0

XauTrop, 2021-01-26:
Thank you for the fast answer. I'm really new in the webthings.I already tryied with the arduinojson_v6.17.0 to v6.17.2, but still some (less) errors:

In file included from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantContent.hpp:11:0,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantSlot.hpp:11,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/MemoryPool.hpp:10,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantData.hpp:7,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/SlotFunctions.hpp:8,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayIterator.hpp:7,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:17,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:23,
                 from C:\Users\Jaume\Documents\Arduino\sketch_jan26a\sketch_jan26a.ino:2:
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantFunctions.hpp: In instantiation of 'bool ArduinoJson6170_91::variantSetInteger(ArduinoJson6170_91::VariantData*, T) [with T = long long int]':
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp:193:32:   required from 'bool ArduinoJson6170_91::VariantRef::set(T, typename ArduinoJson6170_91::enable_if<(ArduinoJson6170_91::is_integral<T>::value && (! ArduinoJson6170_91::is_same<bool, T>::value))>::type*) const [with T = long long int; typename ArduinoJson6170_91::enable_if<(ArduinoJson6170_91::is_integral<T>::value && (! ArduinoJson6170_91::is_same<bool, T>::value))>::type = void]'
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Object/MemberProxy.hpp:46:5:   required from 'typename ArduinoJson6170_91::enable_if<(! ArduinoJson6170_91::is_array<TValue>::value), ArduinoJson6170_91::MemberProxy<TParent, TStringRef>&>::type ArduinoJson6170_91::MemberProxy<TParent, TStringRef>::operator=(const TValue&) [with TValue = long long int; TObject = ArduinoJson6170_91::ObjectRef; TStringRef = String; typename ArduinoJson6170_91::enable_if<(! ArduinoJson6170_91::is_array<TValue>::value), ArduinoJson6170_91::MemberProxy<TParent, TStringRef>&>::type = ArduinoJson6170_91::MemberProxy<ArduinoJson6170_91::ObjectRef, String>&]'
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:317:22:   required from here
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantFunctions.hpp:119:3: error: static assertion failed: To use 64-bit integers with ArduinoJson, you must set ARDUINOJSON_USE_LONG_LONG to 1. See https://arduinojson.org/v6/api/config/use_long_long/
   ARDUINOJSON_ASSERT_INTEGER_TYPE_IS_SUPPORTED(T);
   ^
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantAs.hpp: In instantiation of 'typename ArduinoJson6170_91::enable_if<(ArduinoJson6170_91::is_integral<T>::value && (! ArduinoJson6170_91::is_same<bool, T>::value)), T>::type ArduinoJson6170_91::variantAs(const ArduinoJson6170_91::VariantData*) [with T = long long int; typename ArduinoJson6170_91::enable_if<(ArduinoJson6170_91::is_integral<T>::value && (! ArduinoJson6170_91::is_same<bool, T>::value)), T>::type = long long int]':
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantAs.hpp:94:22:   required from 'T ArduinoJson6170_91::variantAs(ArduinoJson6170_91::VariantData*, ArduinoJson6170_91::MemoryPool*) [with T = long long int]'
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp:256:50:   required from 'typename ArduinoJson6170_91::VariantAs<T>::type ArduinoJson6170_91::VariantRef::as() const [with T = long long int; typename ArduinoJson6170_91::VariantAs<T>::type = long long int]'
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:606:53:   required from here
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantAs.hpp:59:3: error: static assertion failed: To use 64-bit integers with ArduinoJson, you must set ARDUINOJSON_USE_LONG_LONG to 1. See https://arduinojson.org/v6/api/config/use_long_long/
   ARDUINOJSON_ASSERT_INTEGER_TYPE_IS_SUPPORTED(T);

mrstegeman, 2021-01-26:
Are you using the latest version of the webthing-arduino library?

XauTrop, 2021-01-26:
Yes, I already add the library from github by the zup file if the arduino IDE was not taking the last one. I have the 0.12.0

mrstegeman, 2021-01-26:
Try adding this to the top of your source:

```cpp
#define ARDUINOJSON_USE_LONG_LONG 1
```

XauTrop, 2021-01-26:
Adding: #define ARDUINOJSON_USE_LONG_LONG 1
i got:
In file included from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantContent.hpp:11:0,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantSlot.hpp:11,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/MemoryPool.hpp:10,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantData.hpp:7,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/SlotFunctions.hpp:8,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayIterator.hpp:7,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:17,
                 from C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:23,
                 from C:\Users\Jaume\Documents\Arduino\sketch_jan26a\sketch_jan26a.ino:4:
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantFunctions.hpp: In instantiation of 'bool ArduinoJson6172_91::variantSetInteger(ArduinoJson6172_91::VariantData*, T) [with T = long long int]':
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp:191:32:   required from 'bool ArduinoJson6172_91::VariantRef::set(T, typename ArduinoJson6172_91::enable_if<(ArduinoJson6172_91::is_integral<T>::value && (! ArduinoJson6172_91::is_same<bool, T>::value))>::type*) const [with T = long long int; typename ArduinoJson6172_91::enable_if<(ArduinoJson6172_91::is_integral<T>::value && (! ArduinoJson6172_91::is_same<bool, T>::value))>::type = void]'
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Object/MemberProxy.hpp:49:5:   required from 'typename ArduinoJson6172_91::enable_if<(! ArduinoJson6172_91::is_array<TValue>::value), ArduinoJson6172_91::MemberProxy<TParent, TStringRef>&>::type ArduinoJson6172_91::MemberProxy<TParent, TStringRef>::operator=(const TValue&) [with TValue = long long int; TObject = ArduinoJson6172_91::ObjectRef; TStringRef = String; typename ArduinoJson6172_91::enable_if<(! ArduinoJson6172_91::is_array<TValue>::value), ArduinoJson6172_91::MemberProxy<TParent, TStringRef>&>::type = ArduinoJson6172_91::MemberProxy<ArduinoJson6172_91::ObjectRef, String>&]'
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:317:22:   required from here
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantFunctions.hpp:119:3: error: static assertion failed: To use 64-bit integers with ArduinoJson, you must set ARDUINOJSON_USE_LONG_LONG to 1. See https://arduinojson.org/v6/api/config/use_long_long/
   ARDUINOJSON_ASSERT_INTEGER_TYPE_IS_SUPPORTED(T);
   ^
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantAs.hpp: In instantiation of 'typename ArduinoJson6172_91::enable_if<(ArduinoJson6172_91::is_integral<T>::value && (! ArduinoJson6172_91::is_same<bool, T>::value)), T>::type ArduinoJson6172_91::variantAs(const ArduinoJson6172_91::VariantData*) [with T = long long int; typename ArduinoJson6172_91::enable_if<(ArduinoJson6172_91::is_integral<T>::value && (! ArduinoJson6172_91::is_same<bool, T>::value)), T>::type = long long int]':
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantAs.hpp:94:22:   required from 'T ArduinoJson6172_91::variantAs(ArduinoJson6172_91::VariantData*, ArduinoJson6172_91::MemoryPool*) [with T = long long int]'
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp:254:50:   required from 'typename ArduinoJson6172_91::VariantAs<T>::type ArduinoJson6172_91::VariantRef::as() const [with T = long long int; typename ArduinoJson6172_91::VariantAs<T>::type = long long int]'
C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino/Thing.h:606:53:   required from here
C:\Users\Jaume\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantAs.hpp:59:3: error: static assertion failed: To use 64-bit integers with ArduinoJson, you must set ARDUINOJSON_USE_LONG_LONG to 1. See https://arduinojson.org/v6/api/config/use_long_long/
   ARDUINOJSON_ASSERT_INTEGER_TYPE_IS_SUPPORTED(T);
   ^
Found various libraries for "Thing.h"
Usado: C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino
 No usado: C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino-master
Thanks in advance!

mrstegeman, 2021-01-26:
It looks like you have multiple versions of this library installed:
```
Found various libraries for "Thing.h"
Usado: C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino
No usado: C:\Users\Jaume\Documents\Arduino\libraries\webthing-arduino-master
```

XauTrop, 2021-01-26:
Yes, I tried both libraries separately. But still the issues seems that are coming from the arduinojson library

mrstegeman, 2021-01-26:
We already have that `#define` in this library, which is why I wanted to make sure you were using the latest. Can you remove both libraries and just pull in the latest from the Arduino Library Manager? Also, please post your current source.

XauTrop, 2021-01-26:
Fuck, I did a clean start with all the libraries download again an it works!!
Thank you very much!
You do a great work mates!

closed 2021-01-26 by mrstegeman