WebThingsIO/webthing-arduino #132: Arduino device (esp8266) Add events to PushButton

rarno, 2021-02-15:
Hello, please excuse my bad English. I also need help with "PressedEvent". Unfortunately, in the examples "Update LEDLamp example to use actions & events. (# 101)" I cannot find any instructions on "Events". I cannot correctly adapt the "Event" "Overheated" in the example "LEDLamp" to a "pressedEvent". I'm trying a "flip-flop implementation" like in "# 694". Do you have an example for me to learn more?

mrstegeman, 2021-02-15:
Why can you not adapt the LEDLamp example?

rarno, 2021-02-16:
My knowledge is not enough.
Pushed button is pressed, "Pressed Event" comes, information appears on the monitor, an LED can be switched on with it. The next time I press the button, nothing happens. A new event message only comes after I log out and log in again. What I am doing wrong?