WebThingsIO/webthing-arduino #134: Thing Description property has no forms field

JohannesMerkt, 2021-03-03:
I tried to use [node-wot](https://github.com/eclipse/thingweb.node-wot) to connect to an arduino running the example led code from webthing-arduino. It complained that there is no forms field on properties within the thing description. 

Which library isnt following the [Wot-Thing-Description](https://www.w3.org/TR/wot-thing-description/) specification?

It would be very valuable if both libraries could work together.

mrstegeman, 2021-03-03:
Currently, the webthing libraries and the gateway conform to a [draft specification](https://webthings.io/api/), rather than the W3C spec.

closed 2021-03-03 by mrstegeman

benfrancis, 2021-03-04:
See also: https://github.com/WebThingsIO/gateway/issues/2802