WebThingsIO/webthing-arduino #149: LED LAMP Example not working

dizzwold, 2021-11-29:
I don't have much coding knowledge, but Iv'e tried an tinkered with the LED LAMP sketch on a NodeMCU, Wemos D1 and now a Wroom32 (30 pin), with no success.
 With some tinkering I can get it to turn On and OFF, but nothing else.
Anyone else have this problem?

dizzwold, 2021-12-03:
Is there an updated version of the ESP8266 Led Lamp example, i'm getting loads of errors?

NinoBenci, 2022-06-18:
Hi dizzwold.
I've just spotted your issue and noticed that there have been no replies or comments.
What are the errors that you are getting. I've had some success in getting the examples to work in various formats, on Wemos D1 Pro and RobotDroid ESP32 boards.
The majority of issues that I've been having with my WebThings (mostly motion and environmental sensing) has been due to the type of network that I have to connect through, plus some creative tinkering in getting the examples to work.
Let me know if you have managed to make any headway since your post?

dizzwold, 2022-06-18:
Hi Antonio,
 Thank you for coming back to me.
 Sadly, I've given up on Webthings and moved on to Home Assistant, due to the lack of friendly and helpful support. 

 I'm sorry to say this, but I've never been met with so much ignorance with the likes of anything to do with Webthings.
 Thank you anyway, for not being like one of the people I speak of above and replying.
Regards,
 Dizzwold

benfrancis, 2022-06-20:
Hi @dizzwold, I'm sorry that has been your experience.

Please bear in mind that the WebThings project is maintained by a team of volunteers who may not always have time to provide one to one support.

For future reference, you may be more likely to get helpful responses if you share the actual error messages you're seeing and steps for someone else to reproduce them, rather than just a general "something didn't work".

I hope you have a good experience with HomeAssistant, I hear good things about that project.

closed 2022-06-20 by benfrancis