WebThingsIO/webthing-arduino #152: Real-time event notification but without Web Socket

A-R-S-D, 2022-01-31:
hello buddies!
I'm wondering if the Arduino web thing library supports any other method than Web Socket and periodic HTTP Get for event handling that is somehow like webhook? 
For example  when we send "addEventSubscription" message over HTTP
ESP keeps the transmitter's IP then every time an event goes off, iterates over the set of saved IPs and notify every subscribed device with HTTP post/put request. 
Instead of having an open Web Socket connection for each event for a while to rarely receive events.

Thanks in advance!
A. Salehy

benfrancis, 2022-02-01:
Hi @A-R-S-D, no the [Web Thing API](https://iot.mozilla.org/wot/) implemented by this library does not support anything like that.

At some point I expect we'll add support for Server-Sent Events to conform with the W3C WoT [Core Profile](https://w3c.github.io/wot-profile/#protocol-binding), but that will still require keeping a TCP socket open. It does have a mechanism for automatically catching up on missed events though.

The issue with webhooks is that it reverses the client/server roles. This requires every device to be both an HTTP client and HTTP server, but more importantly requires all Consumers to be both an HTTP client and HTTP server as well. That makes implementing a Consumer as a desktop or mobile app very challenging.

One approach we may consider supporting in the future is to use CoAP, which is built on UDP rather than TCP and has an "[observe](https://datatracker.ietf.org/doc/html/rfc7641)" extension.

Another possibility being considered for the [Web Thing Protocol](https://www.w3.org/community/web-thing-protocol/) (standardised version of the WebSocket sub-protocol) is to allow connections to be established from the device side rather than the consumer side, but that again requires consumers to be HTTP servers.

Have you experienced problems with keeping a WebSocket open?

A-R-S-D, 2022-02-12:
Thanks for your response and sorry for my late reply.
I defined a property and clients will put their event callback URL on this property then whenever that event goes off, I send the event to the callback URL using PUT method by HTTPClient library.