WebThingsIO/webthing-arduino #157: Updates to webthing-arduino

Fleatech-Sci, 2025-06-07:
I have forked webthing-arduino and updated Thing.h and ESPWebThingAdapter.h to fulfill the API updates for ArduinoJson 7.x.x, ESPAsyncWebServer 3.7.2 and esp32 3.1.3.

EthernetWebThingAdapter.h and WiFiWebThingAdapter.h are currently a work in progress as I do not have the hardware to test.

The updates to Thing.h and ESPWebThingAdapter are being tested on the esp32-c3 and esp32-s3 at the moment.
These are the only esp32 boards that I have on hand.

If you have the patience, please test on other hardware.
No tests have been performed on esp8266 as yet.

All other information remains the same as in the original webthings-arduino repository.

Repository is at: https://github.com/Fleatech-Sci/webthing-arduino-0.3.0

All inputs and suggestions are welcomed.

benfrancis, 2025-06-11:
Thanks for sharing this. PRs are welcome if you think these updates would be valuable to merge into this repo.

Fleatech-Sci, 2025-06-15:
I'm currently working on updating EthernetWebThingAdapter.h and WiFiWebThingAdapter.h, although I may not be able to test as I do not have the required hardware. I'm also working to update the Seeed WioTerminal version at a later stage.
I'm happy to pass on the updated files to whomever maintains the current repository.
Cheers.